GridProtectionAlliance/gsf #6: Separate connection errors from errorlog

arkrohne, 2016-05-26:
The ErrorLog.txt log is often filled with connection issues of failing or intermittent devices. Separating those out into a second connection error log would simplify finding configuration or operational errors.

ritchiecarroll, 2016-05-26:
How about a flag to optionally log these at all?

arkrohne, 2016-05-26:
That may be the simplest solution. Sometimes it's useful to see if an error occurred near a connect/disconnect event, but you often don't need that data for the whole time.

rmc00, 2016-06-22:
@ritchiecarroll Would this be a flag on the adapter level or something more global? Seems like it could be implemented in InputAdapterBase and OutputAdapterBase on the reconnect timer logic.

ritchiecarroll, 2016-06-22:
@rmc00 that might work, but I think to reduce noise in this case simply implementing some error filter right inside the PhasorMeasurementMapper adapter would suffice. Do you want to take a stab at it?

rmc00, 2016-06-23:
Yeah, I thought I might give it a shot.

ritchiecarroll, 2016-07-15:
Feature completed.

closed 2016-07-15 by ritchiecarroll