GridProtectionAlliance/gsf #14: Adding SQL Server database user for time-series projects still injects SSPI into connection string

ritchiecarroll, 2017-02-21:
When walking through the Configuration Setup Utility (CSU) for a GSF based time-series project, selecting a SQL Server database and creating a new database-level user using integrated authentication for the connection to create the new database and user - the connection string properly adds the user ID and password for the database user to the connection string - but improperly also adds `Integrated Security = SSPI` to the connection string which causes the connection to fail, at least when connecting to a remote database where a local NT SERVICE user will not have access.

This seems to be an issue for all CSU implementations against SQL Server - all CSU implementations will need to be checked, including: openPDC, SIEGate, substationSBG, openHistorian, openMIC, Project Alpha, etc...

GPA-jhyder, 2017-03-08:
Fixed User and SSPI injection into configuration Connection String.

GPA-jhyder

GPA-jhyder, 2017-03-08:
This issue is now closed.

closed 2017-03-08 by GPA-jhyder