GridProtectionAlliance/gsf #17: Possible GSF.Core.Shared.ArrayExtensions.IndexOfSequence Issue

ajstadlin, 2017-03-14:
I'm not sure if this is a problem  or not.  I had to change Line# 460 from original:
```cs
index = Array.IndexOf(array, sequenceToFind[0], index + 1, length - (index + 1));
```
To:
```cs
index = Array.IndexOf(array, sequenceToFind[0], index + 1, array.Length - (index + 1));
```

The following is a link to a quick console app to test.  Test files 1 and 2 work fine.  Test file 3 breaks with an error like:  
```
An unhandled exception of type 'System.ArgumentOutOfRangeException' occurred in mscorlib.dll
Additional information: Count must be positive and count must refer to a location within the string/array/collection.
```

[Link to the test project: Test_ArrayExtensions_IndexOfSequence](https://github.com/ajstadlin/Test_ArrayExtensions_IndexOfSequence)

//AJ

ritchiecarroll, 2017-03-15:
Certainly what we are trying to avoid here is an `ArgumentOutOfRange` exception for the `count` parameter on the `Array.IndexOf` function so that "_startIndex and count specify a valid section in the array_" - this includes avoiding a negative number :-)

The tricky part here is that the `length` parameter for the `IndexOfSequence` function function "should" specify a _portion_ of the source array to consider - and the subsequent sub-searches should never exceed these bounds, hence `length - (index + 1)`.

However, I'm not 100% sure that your test code evokes the spirit of that notion with this line of code:
```c#
findIndex = GSF.ArrayExtensions.IndexOfSequence(fileBytes, findBytes, findIndex + findBytes.Length);
```
Seems like the `length` parameter here should be more like: `fileBytes.Length - (findIndex - 1)`?

If by changing that, does the program behave as expected? If so, we may want to make the documentation of that function more clear on the meaning of the  `startIndex` and `length` parameters.

Thanks for your continued support and testing AJ!

Ritchie

ajstadlin, 2017-03-15:
Thank you for your feedback.  This issue is a challenge for me for sure.

I'm calling the function overload:

```cs
public static int IndexOfSequence<T>(this T[] array, T[] sequenceToFind, int startIndex) where T : IComparable<T>
```
From that function's scope, My code can't see the ```length``` in the overload that uses ```length``` as the parameter.  *However, you might notice that your change suggestion looks a lot like the change I have to make to ArrayExtensions.cs line# 460 to get my use case to work.*

I admit I might be miss interpreting the IndexOfSequence function.   In my specific use case (a one-shot utility), I am going to need to call this function on 81559 files.  I figure by the time I work out all of the exceptions, I will have an intimate understanding for how this function works by the time I am done.  It was merely by chance that my "TestDataFile3.bin" file was the First of the 81559 files. 

//AJ

ajstadlin, 2017-03-15:
If I change ArrayExtensions.cs line# 455 from:
```cs
if (index + x < startIndex + length) 
```
to:
```cs
if ((index + x < startIndex + length) || (length < (index + 1)))
```
The exception is preemptively managed and returns ```index = -1``` when the exception occurs.  This is proving out to be acceptable in my test cases and my use case so far; because the exception only occurs when there are no more occurrences of the sequence being searched for. 

//AJ

ritchiecarroll, 2017-03-15:
Awesome fix! Want to check that in?

StephenCWills, 2017-03-15:
Thanks for reporting this, AJ! Ritchie and I had a closer look at this. Looks like we were calculating the count incorrectly when calling Array.IndexOf(). The change I just checked in should fix this issue.

closed 2017-03-15 by StephenCWills

ajstadlin, 2017-03-15:
Your Fix Works!  I just tested with my use case that calls that function to scan over 81000 files.  100% successful!  

Thank you very much for the Fix!

//AJ

Now I can go home and shovel snow

commit d4a43c6e (2017-03-15): GSF.Core.Shared: Fixed #17 - bug in ArrayExtensions.IndexOf() where it was calculating the wrong count in calls to Array.IndexOf().