GridProtectionAlliance/gsf #23: Feature Request - Windows Performance Counters

EEParker, 2017-06-08:
I would be nice if gsf based applications such as SIEGate, openHistorian, openPDC, etc. had a basic set of windows performance counters, such as the statistics from the main page.

This could be used by standard IT monitoring tools that can utilize windows perf counters for alerting and status monitoring.

[image]

And perhaps optional statistics per input/output streams with totals. 
[image]

StephenCWills, 2017-06-08:
The statistics on the main page are actually all calculated by sampling standard Windows performance counters. You should already have the ability to monitor those using the standard IT monitoring tools you've described.

EEParker, 2017-06-16:
OK, I do think we can get those basic perf counters. 

What do you think about performance counters for the run time statistics for I/O streams? My idea was to have a custom performance counter category for GSF and the counter could have instances for each application+stream. Here is an example.

[image]

ritchiecarroll, 2017-07-20:
I think these could be added fairly easily...

closed 2024-02-23 by EEParker