GridProtectionAlliance/gsf #25: Internal Subscription Security Enhancement

EEParker, 2017-07-19:
We have had several questions about adding a security layer to internal subscriptions, specifically: 

1. Authorization via shared secret or api key
2. TLS with self-signed or internal CA signed certs
3. Publish to client (vs subscribe to server) for higher to lower environment stacks where higher environment is the initiator of the connection.

Does it seem reasonable to add these features to existing applications? The scope of 3 may be fairly large.

ritchiecarroll, 2017-08-01:
1 and 2 are currently supported - but do require that you "allow" points for a specific subscriber - however, you can easily "allow all" so that the subscription works like a normal internal subscription.

Item 3 has been on the list for some time - we figured this would naturally fall out of the ongoing work with STTP. You are correct in that it may be a significant change to the existing code, but like I said, something we've wanted to support for some time.

ritchiecarroll, 2021-01-13:
Item 3 now supported via STTP: https://sttp.github.io/documentation/reverse-connections/

closed 2021-01-13 by ritchiecarroll