GridProtectionAlliance/gsf #38: Support more metadata value substitutions in grafana datasource 'Label' function

mikenugent1, 2017-11-09:
Is it possible to add support for additional metadata substitutions for measurements, such as Description, PhasorLabel, and Phase?  This would allow more flexibility as PointTag often doesn't have enough identifying information for DFR or other devices that have multiple voltage or current phasors under a single device.

ritchiecarroll, 2017-11-10:
Hi Mike,

I posted the code that handles this operation below - we do pull the full meta-data record (i.e., the `DataRow`) for the target tag and support case-insensitive replacement of the `"ID", "SignalID", "PointTag", "AlternateTag", "SignalReference", "Device", "SignalType"` fields when referenced in braces, e.g., `{AlternateTag}` - easy enough to add to the replacement list - have some specific suggestions besides those mentioned above?

```C#
            // Handle label function as a special edge case - groups operations on label are ignored
            if (seriesFunction == SeriesFunction.Label)
            {
                // Derive label
                string label = parameters[0];

                if (label.StartsWith("\"") || label.StartsWith("'"))
                    label = label.Substring(1, label.Length - 2);

                DataSourceValueGroup[] groups = dataset.ToArray();

                for (int i = 0; i < groups.Length; i++)
                {
                    string target = groups[i].RootTarget;

                    string seriesLabel = TargetCache<string>.GetOrAdd($"{label}@{target}", () =>
                    {
                        string derivedLabel = label;
                        DataRow record = target.MetadataRecordFromTag(Metadata);

                        if ((object)record != null)
                        {
                            foreach (string fieldName in s_metadataRecordFields)
                                derivedLabel = derivedLabel.ReplaceCaseInsensitive($"{{{fieldName}}}", record[fieldName].ToString());
                        }

                        // ReSharper disable once AccessToModifiedClosure
                        if (derivedLabel.Equals(label, StringComparison.Ordinal))
                            derivedLabel = $"{label}{(groups.Length > 1 ? $" {i + 1}" : "")}";

                        return derivedLabel;
                    });
                                                              
                    yield return new DataSourceValueGroup
                    {
                        Target = seriesLabel,
                        RootTarget = target,
                        Source = groups[i].Source
                    };
                }
            }

            s_metadataRecordFields = new[] { "ID", "SignalID", "PointTag", "AlternateTag", "SignalReference", "Device", "SignalType" };
```

ritchiecarroll, 2017-11-10:
Could add the whole list I suppose:

https://github.com/GridProtectionAlliance/gsf/blob/master/Source/Documentation/FilterExpressions.md#activemeasurements

ritchiecarroll, 2017-11-10:
Fixed forever - even if available metadata columns change:
```C#
                        DataRow record = target.MetadataRecordFromTag(Metadata);

                        if ((object)record != null)
                        {
                            foreach (DataColumn column in record.Table.Columns)
                            {
                                string fieldName = column.ColumnName;
                                derivedLabel = derivedLabel.ReplaceCaseInsensitive($"{{{fieldName}}}", record[fieldName].ToString());
                            }
                        }
```

See: https://github.com/GridProtectionAlliance/gsf/commit/3ce5773d15fb44b02aef7e6145caa3278ba81987

closed 2017-11-10 by ritchiecarroll

mikenugent1, 2017-11-10:
I like that solution.. thanks Ritchie!

See: 3ce5773<https://github.com/GridProtectionAlliance/gsf/commit/3ce5773d15fb44b02aef7e6145caa3278ba81987>

This email and any attachments are for the sole use of the intended recipient(s) and may contain confidential information. If you receive this email in error, please notify the sender, delete the original and all copies of the email and destroy any other hard copies of it.