GridProtectionAlliance/gsf #39: No rows after query in ApplicationRole from Mysql cause "Failed to load a valid security context"

avs009, 2018-05-17:
Using openPDC manager to access mysql database they are returning on open the message: 

"Failed to load a valid security context - no application roles were found for node ID 'e7a5235d-cb6f-4864-a96e-a8686f36e599', verify the node ID in the config file 'D:\Users\allan.scheid\Documents\PTI\RPMF\OpenPDC\Git\openPDC\Build\Output\Debug\Applications\openPDC\openPDCManager.exe.Config'. Cannot proceed with user authentication."

After debugging, i discovered in AdoSecurityProvider class an error with mysql plugin, as can see in the next image showing the sql query directy on database and they result in c#.

[image]

Versions:
Operational System: Windows 10 (Version 1703, Build 15063.1058)
openPDC: Master from Git
GSF: Master from Git
Mysql client for dotNet: mysql-connector-net-8.0.11-noinstall (local)
Mysql server: 5.5.59 (remote from Debian 8)

StephenCWills, 2018-05-18:
I have the same issue with MySQL Connector/NET 8.0.11. The same exact MySQL database works fine with the version I had previously installed, which was 6.9.8. If you browse their previously generally available versions, they still offer 6.9.12, and I was able to verify just now that it works. I would recommend using that version instead.

avs009, 2018-05-18:
Thank you @StephenCWills!! That's version works successful with openPDC Manager, and solve another error with openPDC in linux!

closed 2018-05-19 by StephenCWills