GridProtectionAlliance/gsf #45: SendDataComplete how bytes are sent?

jenokizm, 2018-09-06:
Hi, im using SendDataComplete event in TcpClient and SendClientDataComplete in TcpServer.
How in both cases to find out how many bytes sent?

ritchiecarroll, 2018-09-06:
For send data complete in TcpClient, you will get the data buffer in argument 1 and the count in argument 2, see below:

https://github.com/GridProtectionAlliance/gsf/blob/master/Source/Libraries/GSF.Communication/ClientBase.cs#L181 

/// <summary>
/// Occurs when data received from the   server has been processed and is ready for consumption.
/// </summary>
/// <remarks>
/// <see cref="EventArgs{T1,T2}.Argument1"/> is a new   buffer containing post-processed data received from the server starting at   index zero.<br/>
/// <see cref="EventArgs{T1,T2}.Argument2"/> is the   number of post-processed bytes received in the buffer from the server.
/// </remarks>
[Category("Data"),
Description("Occurs when data received   from the server has been processed and is ready for consumption.")]
public event EventHandler<EventArgs<byte[], int>> ReceiveDataComplete;

For send data _client_ complete in TcpServer, first arg is a Guid (representing the unique client connection ID) – then arg2 is buffer and arg3 is count, see:

https://github.com/GridProtectionAlliance/gsf/blob/master/Source/Libraries/GSF.Communication/ServerBase.cs#L203

ritchiecarroll, 2018-09-06:
Actually, just noticed you said "Send" not "Receive", sorry - in that case:

jenokizm, 2018-09-07:
Someone from us delicately buggy. But I asked about the events of sending and not receiving. Sending events do not have such agrunments as receiving, this is the problem!

ritchiecarroll, 2018-09-07:
So sorry - my mistake. I am guessing that since these notifications were just indications, i.e., a confirmation, of data that the user had already sent (and hence had access to), that the original buffer was not included in the notification. Do you have a use case where this would be useful?

jenokizm, 2018-09-07:
I need this for a reason. I have a p2p program in each of which the client and the server are running. I want to collect from the sending and receiving events the number of bytes and then show traffic on the graph in real time.
[image]

StephenCWills, 2018-09-10:
FYI, something like this ought to work.

```
        private static void Server_SendClientDataComplete(object sender, EventArgs<Guid> e)
        {
            TcpServer server = sender as TcpServer;
            Guid clientID = e.Argument;
            
            if (server.TryGetClient(clientID, out TransportProvider<Socket> client))
            {
                int bytesSent = client.Statistics.LastBytesSent;
                // Got bytes sent!
            }
        }
```

closed 2018-09-17 by StephenCWills