GridProtectionAlliance/gsf #48: Failing to build

immesys, 2019-01-16:
I am encountering the following build error when building the TimeSeriesPlatformLibrary on master:

```
/gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp: In function 'int32_t GSF::TimeSeries::IndexOf(const string&, const string&, bool)':
/gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:179:46: error: call of overloaded 'distance(std::__cxx11::basic_string<char>::const_iterator, __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >)' is ambiguous
     return distance(value.begin(), it.begin());
                                              ^
In file included from /boost/boost/range/distance.hpp:18:0,
                 from /boost/boost/range/functions.hpp:21,
                 from /boost/boost/range/iterator_range_core.hpp:38,
                 from /boost/boost/lexical_cast.hpp:30,
                 from /boost/boost/multiprecision/detail/number_base.hpp:18,
                 from /boost/boost/multiprecision/detail/default_ops.hpp:10,
                 from /boost/boost/multiprecision/detail/generic_interconvert.hpp:9,
                 from /boost/boost/multiprecision/number.hpp:23,
                 from /boost/boost/multiprecision/cpp_dec_float.hpp:29,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.h:31,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:24:
/boost/boost/iterator/distance.hpp:49:9: note: candidate: constexpr typename boost::iterators::iterator_difference<Iterator>::type boost::iterators::distance_adl_barrier::distance(SinglePassIterator, SinglePassIterator) [with SinglePassIterator = __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >; typename boost::iterators::iterator_difference<Iterator>::type = long int]
         distance(SinglePassIterator first, SinglePassIterator last)
         ^~~~~~~~
In file included from /usr/include/c++/7/bits/stl_algobase.h:66:0,
                 from /usr/include/c++/7/algorithm:61,
                 from /boost/boost/any.hpp:17,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.h:28,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:24:
/usr/include/c++/7/bits/stl_iterator_base_funcs.h:138:5: note: candidate: typename std::iterator_traits<_Iterator>::difference_type std::distance(_InputIterator, _InputIterator) [with _InputIterator = __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >; typename std::iterator_traits<_Iterator>::difference_type = long int]
     distance(_InputIterator __first, _InputIterator __last)
     ^~~~~~~~
/gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp: In function 'int32_t GSF::TimeSeries::IndexOf(const string&, const string&, int32_t, bool)':
/gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:189:46: error: call of overloaded 'distance(std::__cxx11::basic_string<char>::const_iterator, __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >)' is ambiguous
     return distance(value.begin(), it.begin());
                                              ^
In file included from /boost/boost/range/distance.hpp:18:0,
                 from /boost/boost/range/functions.hpp:21,
                 from /boost/boost/range/iterator_range_core.hpp:38,
                 from /boost/boost/lexical_cast.hpp:30,
                 from /boost/boost/multiprecision/detail/number_base.hpp:18,
                 from /boost/boost/multiprecision/detail/default_ops.hpp:10,
                 from /boost/boost/multiprecision/detail/generic_interconvert.hpp:9,
                 from /boost/boost/multiprecision/number.hpp:23,
                 from /boost/boost/multiprecision/cpp_dec_float.hpp:29,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.h:31,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:24:
/boost/boost/iterator/distance.hpp:49:9: note: candidate: constexpr typename boost::iterators::iterator_difference<Iterator>::type boost::iterators::distance_adl_barrier::distance(SinglePassIterator, SinglePassIterator) [with SinglePassIterator = __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >; typename boost::iterators::iterator_difference<Iterator>::type = long int]
         distance(SinglePassIterator first, SinglePassIterator last)
         ^~~~~~~~
In file included from /usr/include/c++/7/bits/stl_algobase.h:66:0,
                 from /usr/include/c++/7/algorithm:61,
                 from /boost/boost/any.hpp:17,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.h:28,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:24:
/usr/include/c++/7/bits/stl_iterator_base_funcs.h:138:5: note: candidate: typename std::iterator_traits<_Iterator>::difference_type std::distance(_InputIterator, _InputIterator) [with _InputIterator = __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >; typename std::iterator_traits<_Iterator>::difference_type = long int]
     distance(_InputIterator __first, _InputIterator __last)
     ^~~~~~~~
/gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp: In function 'int32_t GSF::TimeSeries::LastIndexOf(const string&, const string&, bool)':
/gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:199:46: error: call of overloaded 'distance(std::__cxx11::basic_string<char>::const_iterator, __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >)' is ambiguous
     return distance(value.begin(), it.begin());
                                              ^
In file included from /boost/boost/range/distance.hpp:18:0,
                 from /boost/boost/range/functions.hpp:21,
                 from /boost/boost/range/iterator_range_core.hpp:38,
                 from /boost/boost/lexical_cast.hpp:30,
                 from /boost/boost/multiprecision/detail/number_base.hpp:18,
                 from /boost/boost/multiprecision/detail/default_ops.hpp:10,
                 from /boost/boost/multiprecision/detail/generic_interconvert.hpp:9,
                 from /boost/boost/multiprecision/number.hpp:23,
                 from /boost/boost/multiprecision/cpp_dec_float.hpp:29,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.h:31,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:24:
/boost/boost/iterator/distance.hpp:49:9: note: candidate: constexpr typename boost::iterators::iterator_difference<Iterator>::type boost::iterators::distance_adl_barrier::distance(SinglePassIterator, SinglePassIterator) [with SinglePassIterator = __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >; typename boost::iterators::iterator_difference<Iterator>::type = long int]
         distance(SinglePassIterator first, SinglePassIterator last)
         ^~~~~~~~
In file included from /usr/include/c++/7/bits/stl_algobase.h:66:0,
                 from /usr/include/c++/7/algorithm:61,
                 from /boost/boost/any.hpp:17,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.h:28,
                 from /gsf/Source/Libraries/TimeSeriesPlatformLibrary/Common/CommonTypes.cpp:24:
/usr/include/c++/7/bits/stl_iterator_base_funcs.h:138:5: note: candidate: typename std::iterator_traits<_Iterator>::difference_type std::distance(_InputIterator, _InputIterator) [with _InputIterator = __gnu_cxx::__normal_iterator<const char*, std::__cxx11::basic_string<char> >; typename std::iterator_traits<_Iterator>::difference_type = long int]
     distance(_InputIterator __first, _InputIterator __last)
     ^~~~~~~~
make[2]: *** [CMakeFiles/gsf.dir/Common/CommonTypes.cpp.o] Error 1
```

This is building with boost 1_68_0.

immesys, 2019-01-16:
I believe this is a regression introduced 2 days ago in `b8552be66a8946af418e06ad58ff0a29df4ee35d`

ritchiecarroll, 2019-01-21:
Hi Michael, I am in the process of adding the "Publisher " code to the API - is the error on Linux? as I haven't tested there yet.

immesys, 2019-01-21:
Yeah, the issue is on linux with with GCC 7.3.0 and boost 1.68.0

ritchiecarroll, 2019-02-20:
Michael - Linux should be building now as of this check-in:
https://github.com/GridProtectionAlliance/gsf/commit/bcfcec5aab8bbd2dfbd4afe14186d19513ac942c

Thanks,
Ritchie

closed 2019-02-20 by ritchiecarroll