GridProtectionAlliance/gsf #49: DataSet vs NewDataSet

immesys, 2019-05-17:
I'm using code based on TimeSeriesPlatformLibrary/Samples/AdvancedPublish.cpp which constructs the metadata in the code rather than using an external XML file. When I subscribe I am getting an XML document that looks like this:

```
<?xml version="1.0" standalone="yes"?>
<DataSet>
  <xs:schema id="DataSet" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:ext="urn:schemas-microsoft-com:xml-msdata">
    <xs:element name="DataSet">
      <xs:complexType>
        <xs:choice minOccurs="0" maxOccurs="unbounded">
          <xs:element name="DeviceDetail">
            <xs:complexType>
              <xs:sequence>
etc etc
```

Rather than the metadata I expect and get from OpenPDC which looks like this:

```
<?xml version="1.0" standalone="yes"?>
<NewDataSet>
  <xs:schema id="NewDataSet" xmlns="" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:msdata="urn:schemas-microsoft-com:xml-msdata">
    <xs:element name="NewDataSet" msdata:IsDataSet="true" msdata:UseCurrentLocale="true">
      <xs:complexType>
        <xs:choice minOccurs="0" maxOccurs="unbounded">
          <xs:element name="DeviceDetail">
            <xs:complexType>
              <xs:sequence>
etc etc
```

Notice `DataSet` vs `NewDataSet`. How can I alter the AdvancedPublish example to generate metadata following the NewDataSet schema? I have an external program which is expecting a `NewDataSet`

ritchiecarroll, 2019-05-19:
Line 1475 of `SubscriberConnection.cpp` calls the function which creates the XML:
```
metadata->WriteXml(serializationBuffer);
```

The function being called defines a default parameter for the schema name:
```
void WriteXml(std::vector<uint8_t>& buffer, const std::string& dataSetName = "DataSet") const;
```

To control this, we would need to add a property to `DataPublisher` to set the desired value - however, this makes it even more likely that a user would change this just because they could. My first suggestion would be to have to the external program not depend on the schema name. 

Thanksm
Ritchie

immesys, 2019-05-19:
Ok, I can alter the external program, assuming the actual schema is the same and only the name is the difference

closed 2019-05-19 by immesys