GridProtectionAlliance/gsf #50: foreach ... Remove(..) and "Collection was modified; enumeration operation might not execute."

ajstadlin, 2019-07-08:
First, this isn't really a problem or issue with GSF.Core.  I am posting this issue for insight purposes to better understand a little more about why GSF.Core does Not get this exception.

I have an application that has GSF.Core embedded.  GSF.Core is working fine.  However, in my application, I encountered an interesting exception in a `foreach` routine that calls a `Remove()` function.  I get the "System.Data:  Collection was modified; enumeration operation might not execute." exception in my application call inside a foreach loop.

-=-=-=-  Here is the GSF.Core code I am referring to -=-=-=-

During code review, a noticed a very similar routine in GSF.Core\FileBackedHashSet.cs.

```cs
/// <summary>
/// Removes all elements in the specified collection from the current set.
/// </summary>
/// <param name="other">The collection of items to remove from the set.</param>
/// <exception cref="ArgumentNullException"><paramref name="other"/> is null.</exception>
/// <exception cref="NotSupportedException">The <see cref="FileBackedHashSet{T}"/> is read-only.</exception>
public void ExceptWith(IEnumerable<T> other)
{
    if ((object)other == null)
        throw new ArgumentNullException(nameof(other));

    foreach (T item in other)
        m_lookupTable.Remove(item);
}
```

-=-=-=- Here is my code that results in the exception -=-=-=-
```cs
foreach (DataRow myrow in mytable.Rows)
{
    if (IdontLikeThisRow(myrow))
    {
        mytable.Rows.Remove(myrow);
    }
}
```
-=-=-=-

I appreciate any feedback that will help me better understand this and helps me protect against this exception. FYI: for now, in my application, I simply replaced my foreach() loop with a for() loop as a work around.

//aj

ritchiecarroll, 2019-07-08:
In a word - multi-threading. If multiple threads are accessing a collection, one can remove while another is enumerating - this will cause this exception.

You will either need to synchronize access to the collection with a `lock` or use a thread-safe collection.

ajstadlin, 2019-07-08:
I noticed the .Remove() overloads in the GSF.Core with the Lock and syncroot.   I will look into this.  Thanks!

closed 2019-07-08 by ajstadlin

ajstadlin, 2019-07-08:
For reference:

It looks like my issue is specific to DataRowCollection.Remove() and is documented in Remarks in:
[DataRowCollection.Remove(DataRow) Method](https://docs.microsoft.com/en-us/dotnet/api/system.data.datarowcollection.remove?f1url=https%3A%2F%2Fmsdn.microsoft.com%2Fquery%2Fdev15.query%3FappId%3DDev15IDEF1%26l%3DEN-US%26k%3Dk(System.Data.DataRowCollection.Remove);k(TargetFrameworkMoniker-.NETFramework,Version%3Dv4.6.2);k(DevLang-csharp)%26rd%3Dtrue&view=netframework-4.8)

**"Remove should not be called in a foreach loop while iterating through a DataRowCollection object. Remove modifies the state of the collection. "**

I did not find any DataRowCollection.Remove() like code in GSF.Core.  GSF.Core is great code!  Thank you for making it Open Source!

//aj

StevenChisholm, 2019-07-08:
You're example will always throw this exception. Synchronized or not. The ExceptWith does not throw this exception because the foreach object is not the same as the object that is being modified. 

The easiest fix is to change it to a for loop. However, when you remove an item, don't increment the x on the for loop. Otherwise you'll skip items.

for (int x = 0; x<list.count; )
{
    if (IDontLikeThisItem(list[x])
       list.RemoveAt[x];
    else
        x++;
}

or, I usually iterate the list backwards.

for (int x = list.count-1; x>=0; x--)
{
    if (IDontLikeThisItem(list[x])
       list.RemoveAt[x];
}

ajstadlin, 2019-07-08:
Hi Steven, 

Thank you for the email. I came to the same conclusion - using a for() loop works ok for this in my case. 

The issue caught me off guard since I never had the error before, but have used the Remove() with DataTable.Rows in the past, but now I realize my old code was always looking for only 1 row to remove. 

The following code was working for me for like years, and I never new about this issue. 

foreach (DataRow myrow in mytable.Rows) 
{ 
if (IdontLikeThisRow(myrow)) 
{ 
mytable.Rows.Remove(myrow); 
break; //// this is why my code did not throw the exception in the past. 
} 
} 

When I step by step debug the code without the "break", I hit the exception when the loop continues after the Remove() instead of breaking out. 
I found the Remove()s in the GSF.Core code and wanted to understand why the GSF.Core implementation does not throw the same exception. In other words, make sure someone like me does not contribute a problem like this to the GSF project B-) 

I also like for() loops backwards for removing items. 

Thank you for the feedback. 

//aj