GridProtectionAlliance/gsf #58: .NET Core support

Fedorov-A, 2020-02-25:
Hello.

Is it possible to use GSF libraries (for example COMTRADE) for cross-platform applications? I've checked Gemstone Libraries but there isn't COMTRADE library. Are you planning to port GSF libraries to .NET Core platform? It would be great.

clackner-gpa, 2020-02-25:
Hi Fedorov,
We are planning on moving the COMTRADE GSF library to .Net Core and add it to the Gemstone Libraries in the near future.

mmiecz, 2021-03-03:
I've just stumbled on this issue, and I'm interested in this as well.
What is the current status of .NET Core support for COMTRADE?
Thanks!

ritchiecarroll, 2021-03-03:
We are about to check-in an update to the existing .NET Framework COMTRADE code for support of CFF file format (and other feature updates).

As soon as this is complete (and tested), we will migrate it over to the Gemstone libraries that are based on .NET 5.0 (Core)

blingdaag, 2022-01-24:
Hello, 

We are currently working on a conversion and would like to know if you have any updates you can please share as it relates to your migration to .NET Core (.NET 5 or potentially 6?).

Thanks and Regards,
-jermaine

ritchiecarroll, 2022-01-26:
@blingdaag - does this relate to GSF in general or COMTRADE?

gitkrasty, 2022-01-27:
In my case it is related to COMTRADE specifically. And the decision will have to be taken in a month or two in our team for at least  next 3 years, as it is for a funded project.

ritchiecarroll, 2022-01-27:
Gotcha - this conversion was supposed to happen over the holidays already - so I'll see what I can do about getting it migrated ASAP.

Thanks,
Ritchie

josef-stromsky, 2022-11-21:
Hello,
Any updates in this matter? Is there any estimate, when it could be available? It would be of a great help for us to know the possible date approximately at least.
Thanks a lot.

ritchiecarroll, 2022-11-23:
This is completed: 
* https://github.com/gemstone/comtrade

Docs:
* https://gemstone.github.io/comtrade/

*NuGet:
https://www.nuget.org/packages/Gemstone.COMTRADE

ritchiecarroll, 2023-03-02:
Closing this issue since this has been completed (https://gemstone.github.io/comtrade/)

closed 2023-03-02 by ritchiecarroll