GridProtectionAlliance/gsf #61: Revision of protocols and standards?

JosefLundqvist, 2020-05-26:
Hello,

How do I find out what revision/version is currently implemented of the different standards and protocols in the various libraries?

Regards

ritchiecarroll, 2020-07-08:
Is there a particular standard you would like to know about? Can track it down in the code - not sure there's an official list, there are quite a few protocols supported in the library.

JosefLundqvist, 2020-07-16:
Thank you for the reply!
I'm mostly interested in what revision/edition of IEEE 1159 that is supported.

StephenCWills, 2020-07-16:
IEEE 1159 encompasses more than just a protocol. I'm going to go out on a limb and assume you're interested in the IEEE 1159.3 standard that defines the PQDIF file format used by the GSF.PQDIF library. If so, the latest version of that standard we have on hand is Draft 9 &copy;2002 so the code is based on that version.

If you are also interested in other parts of the standard, then I can tell you that we do adopt some of the standard practices for analyzing and classifying power quality disturbances. However, we don't own any of the standards documents ourselves so these techniques come from information we've gleaned from our colleagues in the industry rather than a specific version of the standard.

JosefLundqvist, 2020-07-22:
Your assumption is correct! Thanks for the information.

closed 2020-07-22 by JosefLundqvist