GridProtectionAlliance/gsf #63: QUESTION: MMS via OSI network protocols

WargodHernandez, 2020-08-24:
Hi I am looking for MMS communication over the original 1990 standards instead of over TCP/IP. 

I had seen the MMS library provided in GSF.MMS and the GSF/Source/Dependencies/OSI

I believe this was the wrong connection to make as the OSI folder seems to be a support for OSIsoft PI Server rather than OSI network protocols. But I wanted to verify that there is no OSI network protocols supported in this library? 

From: [wikipedia - Manufacturing Message Specification](https://en.wikipedia.org/wiki/Manufacturing_Message_Specification#:~:text=This%20version%20of%20MMS%20used%20seven%20layers%20of,implement%2C%20the%20original%20MMS%20stack%20never%20became%20popular.)
[image]

ritchiecarroll, 2020-08-25:
Currently we only have  ASN1 and MMS wire protocols as assemblies, i.e., parser/generator. These tools were added to support IEC protocols, e.g., ICCP - but testing with the protocols mostly focused on TCP/IP wrapped implementations.

WargodHernandez, 2020-08-25:
Thanks for your response, I figured that would be the answer, but wanted to make sure.

Excellent library by the way really well done!

closed 2020-08-25 by WargodHernandez