GridProtectionAlliance/gsf #65: ExpressionEvaluator replacement

gitkrasty, 2020-12-07:
Dear all, 
is there any current plan to get rid or replace of this ExpressionEvaluator package from the future build? Or is its functionality critical and irreplaceable for some GSF features? Its unsupported status since 2016 and the issue with license (#59) is a kind of show stopper for our deployments.
Jan

ritchiecarroll, 2020-12-07:
We do have a replacement in our .NET core code that is Roslyn based: https://github.com/gemstone/expressions

Using this in .NET Framework is possible but brings in about 50 extra DLL's - so we've avoided this.

However the version we are using traces back to the version posted on CodePlex which was MIT-based.

I realize the updated version is no longer MIT licensed, but we use use based on the license at time of use - and have not subsequently upgraded, especially since we have a ready replacement.

Thanks,
Ritchie

gitkrasty, 2020-12-07:
Ahh, thanks a lot for the direction. Works for me. I'll give it a try. Jan

closed 2020-12-07 by gitkrasty