GridProtectionAlliance/gsf #89: Not able to add GFS Nuget packages to Solution

hrvojej, 2021-11-18:
Hi,

I'm trying to add GFS.Core to the empty console project in Visual Studio 2019 and I'm getting:
NuGet Warning NU3037:
https://docs.microsoft.com/en-gb/nuget/reference/errors-and-warnings/NU3037

Any idea how to resolve this?

Cheers!

ritchiecarroll, 2021-11-18:
I do not get this error? What version of GSF are you targeting?

ritchiecarroll, 2021-11-18:
Here's my log:
```
Attempting to gather dependency information for package 'GSF.Core.2.4.2' with respect to project 'GSFNuGetTest', targeting '.NETFramework,Version=v4.6'
Gathering dependency information took 683 ms
Attempting to resolve dependencies for package 'GSF.Core.2.4.2' with DependencyBehavior 'Lowest'
Resolving dependency information took 0 ms
Resolving actions to install package 'GSF.Core.2.4.2'
Resolved actions to install package 'GSF.Core.2.4.2'
Removed package 'GSF.Core.2.4.1-beta' from 'packages.config'
  GET https://api.nuget.org/v3-flatcontainer/gsf.core/2.4.2/gsf.core.2.4.2.nupkg
Successfully uninstalled 'GSF.Core.2.4.1-beta' from GSFNuGetTest
  OK https://api.nuget.org/v3-flatcontainer/gsf.core/2.4.2/gsf.core.2.4.2.nupkg 149ms
Installed GSF.Core 2.4.2 from https://api.nuget.org/v3/index.json with content hash owqlIxUku0+aLwCKEvrXtajacPZ8GghASXDDERhTcsy/8/iCaKzojKoxqOXSoHJT6hobsLxucEAwH/xJ4lEGVA==.
Adding package 'GSF.Core.2.4.2' to folder 'D:\Projects\GSFNuGetTest\packages'
Added package 'GSF.Core.2.4.2' to folder 'D:\Projects\GSFNuGetTest\packages'
Added package 'GSF.Core.2.4.2' to 'packages.config'
Successfully installed 'GSF.Core 2.4.2' to GSFNuGetTest
Removing package 'GSF.Core.2.4.1-beta' from folder 'D:\Projects\GSFNuGetTest\packages'
Removed package 'GSF.Core.2.4.1-beta' from folder 'D:\Projects\GSFNuGetTest\packages'
Executing nuget actions took 2.25 sec
Time Elapsed: 00:00:03.2786832
========== Finished ==========
```

hrvojej, 2021-11-18:
I'm targeting latest version GSF.Core, I guess I'm missing something or something is wrong with my setup.

closed 2021-11-18 by hrvojej