GridProtectionAlliance/gsf #124: gsf (master) git log shows "Apr 5" - like over a week ago

ajstadlin, 2023-04-17:
I'm not sure if this is expected.

I notice the last gsf commit is April 5 instead of like today (April 17).

//AJS

Log below: 

```text
aj@ATMS-DEV MINGW64 /c/GPA/gsf.g (master)
$ git log
commit a9daabed0b883a45f0fc4f57d9411a6cb33cb20d (HEAD -> master, origin/master, origin/HEAD)
Author: StephenCWills <swills@gridprotectionalliance.org>
Date:   Wed Apr 5 11:24:57 2023 -0400

    GSF.Core: Clear watcher thread when FileProcessor is disposed to help prevent ObjectDisposedExceptions

commit e8675a9a5b265d3038d8cfe933a67aa1641305c9
Author: gsfbuildbot <support@gridprotectionalliance.org>
Date:   Sat Apr 15 00:01:28 2023 -0400

    GridSolutionsFramework: Version change for build v2.4.130.0-master.

commit 48cc30266c817080869d5ab11b2bc0d3bad4cfd0
Author: J. Ritchie Carroll <ritchiecarroll@gmail.com>
Date:   Fri Apr 14 11:08:16 2023 -0500

    GSF.Web: Updated msalConfig redirectUri to a fixed value of "/" for simplicity in Azure AD configuration

commit 80812b23cde9c8bd88af9d0230c68b90d5656382
Author: J. Ritchie Carroll <ritchiecarroll@gmail.com>
Date:   Fri Apr 14 09:52:11 2023 -0500

    GSF.Web: Updated property restore on POSIX platforms for AzureAD

```

StephenCWills, 2023-04-18:
We've updated our internal practices to encourage having our changes reviewed by peers before they are merged into the master branch. This commit was indeed made on April 5 and then rebased to the top of the commit history today when the pull request was merged. See #123.

StephenCWills, 2023-04-18:
As an aside, you might try `git log --pretty=fuller` to see more details about the commits. You can see the date the commit was authored as well as when it was actually committed to the branch during the merge.

```
commit a9daabed0b883a45f0fc4f57d9411a6cb33cb20d (HEAD -> master, origin/master, origin/HEAD)
Author:     StephenCWills <swills@gridprotectionalliance.org>
AuthorDate: Wed Apr 5 11:24:57 2023 -0400
Commit:     J. Ritchie Carroll <ritchiecarroll@users.noreply.github.com>
CommitDate: Mon Apr 17 14:28:49 2023 -0500

    GSF.Core: Clear watcher thread when FileProcessor is disposed to help prevent ObjectDisposedExceptions
```

ajstadlin, 2023-04-18:
Thank you for the clarification.  This was the first time I noticed a backward date in the log.

I appreciate the tip and all of your hard work on this project.

//AJS